euratom-software/calcam #19: gui run bug

hagorms, 2018-03-15:
after installation and running gui, no buttons are active (opens and closes a window with no log/output) 
[image]

hagorms, 2018-03-15:
It also happens, if I do start gui from the console/pycharm

ssilburn, 2018-03-17:
Hi, please can you let me know what versions of Python, VTK and PyQt you're using, and / or what Python distribution?

ssilburn, 2018-03-18:
Hi, this was due to a bug in some exception handling code (probably a hangover related to Python 2 / 3 compatibility).

It should now be fixed in commit  ea1aa9a  - please pull the latest version from master and try it again.

hagorms, 2018-03-20:
Hi. Thank you. I was at the conference for these 2 days, so i didnt have an
access to working machine. Thank you, i will test tmr

hagorms, 2018-03-21:
Closed. Well done. Thank you!

closed 2018-03-21 by hagorms