euratom-software/calcam #37: Font and background color are both white in Mac-Os

nick-walkden, 2019-02-07:
Both the font and background color are white in selection menus, see image.

[image]

ssilburn, 2019-02-19:
Hi Nick, (I know we spoke about this but wanted to write it down here). As far as I can tell this is due to a general problem with Qt and OSX dark theme; the font colours etc are not explicitly set by Calcam. From some research it looks like it should be fixed in Qt 5.12, but I'll leave this issue open for now in case it isn't. If you get PyQt 5.12 please report back whether that fixes this not!

ssilburn, 2019-11-06:
Hi Nick, is this still happening?

ssilburn, 2021-04-15:
In the absence of any evidence this is still happening or any way for me to test it, because this has been sitting here idle for ages, I'm going to close this issue. I'll keep the warning in the documentation that this happens with PyQt < 5.12, hopefully it was actually fixed in 5.12 but I have no way of checking.

closed 2021-04-15 by ssilburn