euratom-software/calcam #48: GUI doesn't update until next interaction

TomFarley, 2019-11-07:
In both the point picker and CAD definition GUIs, elements don't update until the next GUI interaction. 

Eg. loading of images, CAD models, image enhancements, zooming in CAD and creation/movement of control points don't update until a different tab is clicked, another control point is placed, or another checkbox/button is pressed.

Version numbers:
Calcam: 2.4.0 on win32 using Enthought Canopy 2.1.9.3717
VTK: 7.0.0
OpenCV: 4.1.1
PyQT: 5.13.1

ssilburn, 2019-11-07:
This doesn't seem to happen with VTK 8.1.2 and PyQt 5.11.2, so likely a change in PyQt and/or VTK between versions which is causing this - needs investigating.

ssilburn, 2019-11-07:
Hmm, so far I haven't been able to reproduce this on my Windows 10 x64 PC with either Canopy 2.1.6.something or 2.1.9.3717, with PyQt 5.11, 5.13.1 or 5.13.2, OpenCV 4.0.0 or 4.1.1, VTK 7.0, 7.1.1 or 8.1.2. So I don't know :P

ssilburn, 2019-12-03:
Closing this because it turned out to be some sort of specific compatibility issue - fixed in this case by upgrading to VTK 8.1.2, so seems to be related to certain combination(s) of VTK version and graphics setup.

closed 2019-12-03 by ssilburn