euratom-software/calcam #56: get_ray_end raises exception with coords='Original'

TomFarley, 2020-01-15:
When calling `get_ray_data` with no pixel coordinate arguments in order to return the ray end coordinates for the whole image as follows:

`surface_coords = ray_data.get_ray_end(coords='Original')`

the following exception is raised when coords='Original':

`File "/home/tfarley/repos/calcam/calcam/raycast.py", line 609, in get_ray_end
    return self.transform.display_to_original_image(self.ray_end_coords)
  File "/home/tfarley/repos/calcam/calcam/coordtransformer.py", line 215, in display_to_original_image
    raise Exception('Expected (multiple of) {:d}x{:d} pixel image, got {:d}x{:d}!'.format(expected_size[0],expected_size[1],image.shape[1],image.shape[0]))
Exception: Expected (multiple of) 256x320 pixel image, got 320x256!`

This exception is not raised and the function performs as expected if either specific pixel coordinates are supplied, or `coords='Display'` i.e. the error only occurs when returning the full frame of 'Original' ray end coordinates. 

Calcam: v2.4.1

ssilburn, 2020-01-15:
Seems my instsance on indexing arrays [y,x] while thinking of image shapes in (x,y) has finally caught up with me ;). Thanks for reporting, I'll check it out.

ssilburn, 2020-01-26:
Something weirder than I thought is going on here, I can't reproduce this with one of my calibrations. This: 

>print(calcam.__version__)
rd = calcam.RayData('test_scratch_raydata.nc')
print(rd.transform.get_display_shape())
print(rd.transform.get_original_shape())
test = rd.get_ray_end(coords='display')
print(test.shape)
test2 = rd.get_ray_end(coords='original')
print(test2.shape)

Produces:

> 2.4.1
(488, 512)
(512, 488)
(512, 488, 3)
(488, 512, 3)

and no exceptions. Could you please send the calibration or ray data file so I can try it with exactly the same data?

ssilburn, 2020-08-02:
I actually came across a very similar error which I think is probably caused by the same bug this week - it's now fixed on the master branch.

closed 2020-08-02 by ssilburn

TomFarley, 2020-12-07:
Sorry I missed your message back in January. Glad you got to the bottom of it in the end.