euratom-software/calcam #58: Inaccurate results from render_cam_view() on VTK versions 8.2+

ssilburn, 2020-07-21:
calcam.render_cam_view() currently gives inaccurate results if using VTK 8.2.0 or newer. This makes it extremely difficult to do calibrations using the GUI because CAD wireframe overlays, which are produced using render_cam_view(), are not accurate and therefore misleading. First reported by Jordan Cavalier, confirmed on both Linux & Windows up to VTK 9.0.1.

ssilburn, 2020-07-21:
I've opened an issue about this against VTK: https://gitlab.kitware.com/vtk/vtk/-/issues/17964

ssilburn, 2021-04-05:
I think there's a solution! Thanks to: http://vtk.1045678.n5.nabble.com/Offscreen-rendering-beyond-screen-resolution-with-WindowToImageFilter-magnification-td5724102.html . It looks like the answer is simply to use SetBorders(0) on the offscreen render window and then not use vtkRenderLargeImage at all.

TomFarley, 2021-04-07:
Great! Does this mean the latest version of vtk is safe for Calcam now? (I'm about to request a python 3.9 distribution on freia with an appropriate version of vtk.)

ssilburn, 2021-04-07:
Not the version on master yet, but very nearly...I'm working on it on a new branch called `offscreen_render_dev`. In a fever of optimism and ambition I took the approach of not just implementing the minimum laziest fix but trying to keep the code nice and clean at the same time....and now benchmarking that branch against master shows the wireframe alignment is coming out slightly different again, but I think that's a new bug which is a side-effect of what I've done.

I'm confident enough that the underlying problem is fixed that I'd say go ahead and request up-to-date VTK for the new python env, but if you want to hang on until I can confirm it's really fixed I'll try to look at it over the next couple of days.

TomFarley, 2021-04-07:
Ok great, good to know. I'll go ahead and request the latest version then. Thanks Scott!

ssilburn, 2021-04-08:
OK, as above - now merged on to master, the code on master tests OK for me on "old" VTK < 8.2 and also on VTK 9.0.1. I'm going to (finally) close this issue! 

After I've delt with a few more issues I'll do a new release for this which will be Calcam 2.7 :)

closed 2021-04-08 by ssilburn