euratom-software/calcam #59: Unable to find CAD model (.ccm file)

dcambonsilva, 2020-07-30:
When trying to open the gui calibration tool a window appears . This bug happens in both versions 2.4.1 and 2.5.0.
﻿"This tool uses CAD models but you seem to have no CAD models set up in calcam. ﻿You can either browse for a folder containing existing Calcam CAD definition (.ccm) files, or create a new CAD model definition."
Choosing "Browse": it is unable to recognise .ccm files : "The selected directory does not seem to contain any CAD definition files. The application will now close."

Choosing "Create new": ﻿it opens a CAD model definition editor window, from there I try to open the MAST.ccm file however the following exception appears: "﻿An unhandled exception has been raised; the action you were performing may have partially or completely failed. This is probably a bug in Calcam; to report it please save an error report file and report the problem at here and/or consider contributing a fix!"

DIAGNOSTIC INFORMATION
----------------------

Platform:       darwin
Python version: 3.7.1 (default, Dec 14 2018, 13:28:58) 
[Clang 4.0.1 (tags/RELEASE_401/final)]
Calcam version: 2.5.0
VTK version:    9.0.0
OpenCV version: 3.4.5
PyQt version:   5.9.6

Traceback (most recent call last):
  File "/Users/davidcambonsilva/Desktop/PhD-CDT-3/calcam_2/calcam/calcam/gui/cad_edit.py", line 79, in <lambda>
    self.action_open.triggered.connect(lambda: self.open())
  File "/Users/davidcambonsilva/Desktop/PhD-CDT-3/calcam_2/calcam/calcam/gui/cad_edit.py", line 626, in open
    self.cadmodel = CADModel( filename , status_callback = self.update_cad_status)
  File "/Users/davidcambonsilva/Desktop/PhD-CDT-3/calcam_2/calcam/calcam/cadmodel.py", line 95, in __init__
    with self.def_file.open_file( 'model.json','r' ) as f:
  File "/Users/davidcambonsilva/Desktop/PhD-CDT-3/calcam_2/calcam/calcam/io.py", line 188, in open_file
    raise IOError('File "{:s}" not in here!'.format(fname))
OSError: File "model.json" not in here!

ssilburn, 2020-07-31:
Hi David,

The reason it claims not to find the .ccm file is almost definitely the same error you're getting when trying to open it in the editor. The question is whether this is a problem with Calcam or your copy of MAST.ccm has got broken somehow. The .ccm files are just zip files with a different extension - can you open MAST.ccm with a zip program and check what's in there? There should be 3 files: "model.json" , "usercode.py" and "wall_contour.txt", and a folder called ".large". Calcam is getting upset because it can't read "model.json", so you need to check if it's really missing or Calcam just isn't seeing it. Here are the contents of my copy of MAST.ccm:
[image]

dcambonsilva, 2020-08-03:
Hi Scott,
On macos it cannot open it to see the contents.
I replaced my .ccm with a fresh version and now it's working again. 
I don't know why my old .ccm file was corrupted, sorry I can't help to identify this bug.
Many thanks.

ssilburn, 2020-08-03:
OK no problem. Let me know if it happens again, maybe there's something specific to OSX + Calcam which managed to corrupt the file so it'd be interesting to see if it was a one-off or not (I don't have an OSX machine to test on unfortunately).

closed 2020-08-03 by ssilburn