euratom-software/calcam #60: Subview_mask not reset when detector_window reset

TomFarley, 2020-09-04:
## Application
I am trying to temporarily 'unapply' a detector_window (after having used it to correctly apply image transformations etc) to make comparisons to the full (non-subwindowed) calibration image stored in the calibration. 

## Error
However when I perform `calcam_calib.set_detector_window(None)` the subview mask(s) is not reset to its original size so that when I subsequently call `calcam_calib.get_subview_mask(coords='Display')` an exception is raised on [line 259 of coordinatetransformer.py](https://github.com/euratom-software/calcam/blob/1b8692b3bb90013c615f389417cd7c4e654d9975/calcam/coordtransformer.py#L259).

# Apparent cause
This appears to be due to the call to `self.geometry.get_original_shape()` on [line 487 of calibration.py](https://github.com/euratom-software/calcam/blob/1b8692b3bb90013c615f389417cd7c4e654d9975/calcam/calibration.py#L487-L489) returning the same (current) subview mask, that is then passed to `self.set_subview_mask` leading to no change. This seems to be a consequence of `x_pixels` and `y_pixels` values not being switched back to the 'native' values until [line 518](https://github.com/euratom-software/calcam/blob/1b8692b3bb90013c615f389417cd7c4e654d9975/calcam/calibration.py#L518) (ie after the subview mask is updated).  
Therefore some refactoring of the order the changes are applied is probably required to fix this, but I'm not familiar enough with the intentions behind how the old values are used in the intervening lines above to know how to fix this safely without other (perhaps subtle) unintended consequences!

ssilburn, 2020-12-05:
Hi Tom,

Sorry it took me so long to get to this! And thanks for the detailed & specific bug report :). This is fixed as of commit d9643bd - I've changed the logic so now it always just takes a copy of the version before any cropping and restores that instead of generating a new mask based on the dimensions from get_original_shape(), which fixes this and also makes things a bit simpler.

closed 2020-12-05 by ssilburn

TomFarley, 2020-12-07:
Awesome, thanks Scott!