euratom-software/calcam #65: Smaller images appearing in calibration windows

rdoyle45, 2021-01-07:
A smaller window has begun appearing in either the CAD window or the CIS image window (shown in screenshot) when I add chessboard images to the calibration.

Once the smaller image in present any input on the parent image is shown in the smaller image.

Not sure if this is a bug or if I am doing something incorrectly.

https://user-images.githubusercontent.com/48983334/103906876-69149480-50f8-11eb-887b-113bb39f9ed8.mp4

[image]

ssilburn, 2021-01-07:
That's very weird, haven't seen that one before! Please can you go in the Calcam Settings GUI and send / attach the info about what versions of everything you have? Then I can see if I can manage to reproduce it here.

rdoyle45, 2021-01-07:
Just noticed Issue #58, I'm going to downgrade VTK and see if that changes anything.

[image]

ssilburn, 2021-01-07:
Aha, OK yeah it would be very interesting to know if this is another VTK version related bug, if it is that might also explain why I haven't seen it. Even if it doesn't fix this particular problem, definitely stay with VTK <8.2 because otherwise the overlay to show you how good the calibration is can come out wrong.

Let me know if downgrading VTK fixes it, and if so I can add this to the list of VTK version compatibility problems that need fixing when I have time.

rdoyle45, 2021-01-07:
It does appear to be related to VTK as I can not replicate it now.

ssilburn, 2021-01-08:
OK cool. I'm actually going to leave this open though so it's visible as something else that needs fixing or at least checking with newer VTK versions.

ssilburn, 2021-04-15:
I still don't really understand the cause of this but I've been able to reproduce and fix it (with a hacky workaround) - now on master, will be released in v2.7 which will be fully compatible with VTK 9.

closed 2021-04-15 by ssilburn