euratom-software/calcam #67: Wire frame overlay returns black image

TomFarley, 2021-03-10:
I have never been able to generate a wire frame view of my calcam calibrations, with the resulting image always coming out as all black (ie all zeros). (Occasionally I have got a crazy patch work of bayer filter like colours). However, this would be really useful to get working now.

I've tried following the flow used in the calcam gui to generate the wire frame image:
```python
    orig_colours = cad_model.get_colour()
    cad_model.set_wireframe(True)
    cad_model.set_colour((0, 0, 1))
    overlay = calcam.render_cam_view(cad_model, calcam_calib, transparency=True, verbose=False, aa=2)
    cad_model.set_colour(orig_colours)
    cad_model.set_wireframe(False)
```

I've also tried a few other things like:
```python
    cad_model.set_flat_shading(False)  # lighting effects
    cad_model.set_wireframe(True)
    cad_model.set_linewidth(3)
    color = cad_model.get_colour()
    cad_model.set_colour((0.5, 0.5, 0.5))
```

I'm working with the MAST-U CAD and a calibration such as `calcam_calib=calcam.Calibration(load_filename='/home/tfarley/calcam2/calibrations/rit_43141_218_nuc_eq_glob-v1.ccc'`

Can you see anything I've missed or should try to fix this?

ssilburn, 2021-03-10:
Hi Tom, hmmm it looks like this might be some new and interesting compatibility problem :(. I downloaded the calibration file you named there, and ran exactly your code block (the top one) on my home Windows PC, and it works as expected - I get this:
[image]
 Since I can't immediately reproduce this on my system, my guess is it must be some sort of so far un-discovered library version or system compatibility thing. Please can you let me know what OS and versions of python / opencv / VTK you're having this on?

TomFarley, 2021-03-15:
Hi Scott, ahh that makes sense that it's a package compatibility issue. Thanks for testing the code. I have the problem running my analysis on freia which I'm aware already has compatibility problems for displaying cad in the GUI. Perhaps the wire frame drawing issue is due to the vtk problem?
[image]
We really should be able to run calcam with all it's functionality on freia. Do you know what the minimum set of package updates required on freia is and I'll get campaigning for them to be installed?

ssilburn, 2021-03-15:
Ah, yes that'll be it then - the code path for calcam.render_cam_view() does use the same VTK rendering routines as the GUI so it'll be the same problem. 

I don't think it's an issue with the versions as such, those versions should be fine with it as far as I know - I think the VTK build/install on Freia (and Heimdall) is just broken. I don't know if it's the same on Freia but on Heimdall it behaves differently on different Python versions. I've taken a step back and tested with a basic VTK python example: https://kitware.github.io/vtk-examples/site/Python/GeometricObjects/CylinderExample/ . On Heimdall, under python 3.5.1 this example code works fine, and so does Calcam. On Python 3.7, where Calcam just gives all black screens / renders, the VTK example code throws up some OpenGL related exceptions in the console window and the VTK window doesn't work. So I think it must be something to do with the way VTK is built/installed. I seem to remember when they were creating the Python 3.7 distribution Alex Meakins struggled with this and no one ever worked out how to to get it work properly.

TomFarley, 2021-03-15:
Ok, interesting. So avenues to explore are:
* See if freia admins can update OpenGL drivers to fix problem
* Try to simplify the VTK installations under (hopefully soon to be installed) python 3.9 with only one version in py3.9 environment
* Try to revert to using VTK7 for calcam on freia

As far as I can tell, only vtk8.1 is installed on freia for python 3.7. If I could get vtk7.x somehow installed alongside, would there be a way of requesting calcam to try importing that version? Thanks!

ssilburn, 2021-03-15:
I'm not sure how it works with VTK versions, since it's a C++ library and then a python wrapper rather than a pure python package so I don't know what other files in what locations the python wrapper will be hooking in to. I assume like any other package Calcam/Python should use whichever comes first in $PYTHONPATH, so if you could get both installed and make sure 7 comes before 8 in Python's search path probably it would work.

TomFarley, 2021-03-15:
Ok, that sounds like something to try. I'll see if I can get a freia sysadmin's opinion on this. 
I just tried running calcam on freia's python 3.5 which appears to have vtk7.1 to see if that works and find that python immediately sef faults on importing calcam :P. Is that a known issue?

ssilburn, 2021-03-15:
Ooof, no that's a new one. Just tried it as me on python 3.5 on Freia 22 and that doesn't happen, it seems to all work fine including VTK displaying the CAD correctly! My module list looks like:

Currently Loaded Modulefiles:
  1) standard/2014-08-12            12) matlab/14(default)
  2) dot                            13) ifort/12.0(default)
  3) idl/08.3(default)              14) mast_specview/2.2
  4) ida/3.21                       15) ipx/1.0.2
  5) uda/2.4.0(default)             16) texlive/2018
  6) uda-mast/1.3.1(default)        17) openmpi/1.8.2
  7) mastcodes/1.0.3                18) nag/Mark23
  8) mastu_idl_data_analysis/1.0.3  19) FUN/1.9
  9) xpad/V8.2.2                    20) vtk7/3.5.1
 10) java/1.8                       21) python/3.5
 11) mdsplus/6.1

TomFarley, 2021-03-15:
tfarley@freia022:~$ module list -t
Currently Loaded Modulefiles:

standard/2014-08-12
dot
idl/08.3(default)
ida/3.21
uda/2.4.0(default)
uda-mast/1.3.1(default)
mastcodes/1.0.3
mastu_idl_data_analysis/1.0.3
xpad/V8.2.2
java/1.8
mdsplus/6.1
matlab/14(default)
ifort/12.0(default)
mast_specview/2.2
ipx/1.0.2
texlive/2018
openmpi/1.8.2
nag/Mark23
FUN/1.9
use.own
vtk7/3.5.1
**python/3.5**

tfarley@freia022:~$ echo $PYTHONPATH
/usr/local/depot/python-3.5.1-**vtk-7.1.0**/lib/python3.5/site-packages:
/usr/local/depot/mdsplus-6.1/mdsobjects:
/usr/local/depot/uda-mast-1.3.1/python:/usr/local/depot/uda-2.4.0/python:

tfarley@freia022:~$ python
Python 3.5.1 (default, Apr 29 2016, 10:11:48) 
[GCC 4.8.3 20140911 (Red Hat 4.8.3-9)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import calcam
**Segmentation fault**

So apart from adas our PYTHONPATHs look the same and the order is different, but I think our modules match too.

ssilburn, 2021-03-15:
Baffling. The only other possibility I can think of is $LD_LIBRARY_PATH affecting VTK maybe?

[image]

(sorry I didn't manage to format my paths in a more readable way)

TomFarley, 2021-03-16:
That doesn't seem to be it. Our $LD_LIBRARY_PATHs are identical and the other python imports work fine.

[image]

Mysterious....

ssilburn, 2021-03-17:
Hmm, yep frankly I'm completely out of ideas. I even tried running your .bashrc to make sure the environment was as similar as possible and it still imports fine for me.

sgibson-mse, 2021-03-24:
Hi All,

Probably not a permanent fix but...

I had this issue too, but found a combination of **python/3.7** and the **vtk7/3.5.1** module worked and I could see the CAD models (And didn't segfault) My full module list looks like:

>module list

Currently Loaded Modulefiles:

>   1) idl/08.3(default)              11) ifort/12.0(default)
>   2) ida/3.21                          12) mast_specview/2.2
>   3) uda/2.4.0(default)           13) ipx/1.0.2
>   4) uda-mast/1.3.1(default)  14) texlive/2018
>   5) mastcodes/1.0.3              15) openmpi/1.8.2
>   6) mastu_idl_data_analysis/1.0.3  16) nag/Mark23
>   7) xpad/V8.2.2                    17) FUN/1.9
>   8) java/1.8                          18) vtk7/3.5.1
>   9) mdsplus/6.1                    19) python/3.7
>  10) matlab/14(default)

TomFarley, 2021-03-24:
Nice spot Sam! I can confirm the vtk7/3.5.1 module in place of vtk-7.1.0/ works for me too. Hadn't spotted that module with the alternative naming. I'll be sticking that in my bashrc :)
The vtk module switch doesn't fix the seg fault on python/3.5, but I don't need to use 3.5, so that's not a problem for me.

ssilburn, 2021-04-03:
Great! I'll close this then since it was an environment issue and not inside calcam.

closed 2021-04-03 by ssilburn

sgibson-mse, 2021-06-04:
ARGH sorry to re-open this again however... python/3.7 and vtk7/3.5.1 modules are resulting in black screens again for me - can anyone else confirm? I'm using nomachine for a virtual desktop, could it be that?

ssilburn, 2021-06-04:
I don't remember having any issues with NoMachine with this before. It's working normally for me both on Freia via VNC and on Heimdall via NoMachine with python/3.7 and vtk/3.5.1. Somehow if I unload vtk/3.5.1 on Heimdall I can still open Calcam and then I get a black screen, but if I try unloading VTK on Freia then Calcam won't open at all. So I can find a situation that reproduces it but I'm not quite sure how :P