euratom-software/calcam #78: Problems with VTK 9+

ssilburn, 2022-02-06:
Using VTK newer than 9.0.1 can cause crashes (python terminates without warning) when setting CAD model to wireframe (e.g. for rendering wireframe overlays for calibration checking). This seems to be an issue in VTK with vtkFeatureEdges trying to allocate huge amounts of memory and failing, so I don't think there's much we can do about it except to stick to older versions and hopefully it will be fixed in VTK. Seems to be the same problem as described in https://discourse.vtk.org/t/handling-allocation-errors-in-python/7451

ssilburn, 2022-06-26:
Noting that the field of view display going invisible was actually a different bug (#86) which is now fixed.

perek42, 2022-07-28:
This issue comes back in the fitting_calib.py by crashing via segmentation fault on an M1 2020 Mac with 10.4 

commenting the line 207 "self.overlay_opacity_slider.setValue(self.config.main_overlay_colour[3]*100)" prevents the crash.

pip list 
Package                 Version
----------------------- ---------
absl-py                 1.1.0
aiohttp                 3.8.1
aiosignal               1.2.0
astunparse              1.6.3
async-timeout           4.0.2
attrs                   21.4.0
autobahn                22.6.1
Automat                 20.2.0
brotlipy                0.7.0
cachetools              5.2.0
calcam                  2.9.1
certifi                 2022.6.15
cffi                    1.15.0
charset-normalizer      2.0.12
click                   8.1.3
colorama                0.4.4
conda                   4.13.0
conda-package-handling  1.8.1
constantly              15.1.0
cryptography            37.0.2
cycler                  0.11.0
flatbuffers             1.12
fonttools               4.34.4
frozenlist              1.3.0
gast                    0.4.0
google-auth             2.9.0
google-auth-oauthlib    0.4.6
google-pasta            0.2.0
grpcio                  1.42.0
h5py                    3.6.0
hyperlink               21.0.0
idna                    3.3
importlib-metadata      4.12.0
incremental             21.3.0
joblib                  1.1.0
keras                   2.9.0
Keras-Preprocessing     1.1.2
kiwisolver              1.4.4
libclang                14.0.1
Markdown                3.3.7
matplotlib              3.5.2
multidict               6.0.2
numpy                   1.22.3
oauthlib                3.2.0
opencv-python-headless  4.6.0.66
opt-einsum              3.3.0
packaging               21.3
Pillow                  9.1.1
pip                     22.1.2
protobuf                3.19.4
pyasn1                  0.4.8
pyasn1-modules          0.2.8
pycosat                 0.6.3
pycparser               2.21
pydantic                1.9.1
pyOpenSSL               22.0.0
pyparsing               3.0.9
PyQt6                   6.3.1
PyQt6-Qt6               6.3.1
PyQt6-sip               13.4.0
PySocks                 1.7.1
python-dateutil         2.8.2
PyYAML                  6.0
requests                2.27.1
requests-oauthlib       1.3.1
rsa                     4.8
ruamel-yaml-conda       0.15.80
scipy                   1.8.1
setuptools              62.3.2
six                     1.15.0
tensorboard             2.9.1
tensorboard-data-server 0.6.1
tensorboard-plugin-wit  1.8.1
tensorflow-addons       0.17.1
tensorflow-estimator    2.9.0
tensorflow-macos        2.9.2
tensorflow-metal        0.5.0
termcolor               1.1.0
tqdm                    4.64.0
Twisted                 22.4.0
txaio                   22.2.1
typeguard               2.13.3
typer                   0.4.1
typing_extensions       4.2.0
urllib3                 1.26.9
vtk                     9.0.3
Werkzeug                2.1.2
wheel                   0.37.1
wrapt                   1.14.1
wslink                  1.6.6
yarl                    1.7.2
zipp                    3.8.0
zope.interface          5.4.0

ssilburn, 2022-11-28:
This is now fixed (at least for now) in version 2.10.

closed 2022-11-28 by ssilburn