euratom-software/calcam #83: Suggestion to upload the calcam to PyPI.

munechika-koyo, 2022-03-13:
I would suggest uploading this package to Python Package Index in order to install it with the `pip` command.
I think `calcam` is a very useful package and use it in my original one, so t is more preferable to list `calcam` as a dependent package and manage it with `pip`.

What would you think of this idea?

I would be able to help with this modification using like `setup.cfg`, `pyproject.toml`, etc.  if necessary.

Thank you in advance.

ssilburn, 2022-06-13:
Hi! Yes, originally I didn't package it for PyPI because the dependencies were very tricky so it didn't make much sense, but more recently the dependencies are mostly easily available on PyPI also so this should be more feasible! If you have time and know how to do this, feel free to go for it and put in a pull request! Otherwise I can have a look at it for the next update.

munechika-koyo, 2022-07-15:
Thank you for welcoming my contribution.
I am developing the packaging system according to [setup.cfg config.](https://setuptools.pypa.io/en/latest/userguide/declarative_config.html)

In addition, I am aiming at producing wheel file and upload to pypi.

ssilburn, 2022-07-15:
Hi,
Thanks! Although I already mostly worked on this - if you look at the head of the master branch, I have made setup.py suitable to build source distributions.

I already uploaded it to TestPyPi - https://test.pypi.org/project/calcam/

When 2.9.0 is done (just a few bugs left to fix), I was planning to put this source distribution on PyPi. If you manage to get it to build wheels that will work across a range of platforms I'm happy to add that in, but I couldn't think of a way to properly handle the dependencies, e.g. requiring EITHER opencv-python OR opencv-python-headless, and EITHER PyQt4, PyQt5 OR PyQt6. So my thought was, distributing it as source is the best way to make it easy to install on the widest range of platforms.

munechika-koyo, 2022-07-15:
I am just wondering why we do not choose one option for such a dependency.
Is it not allowed to set specific dependencies for example "open-python-headless" and "PyQt6"?

munechika-koyo, 2022-07-15:
Thank you for your kind reply.
I understand there are a lot of platforms to use calcam.
(However Canopy already reached End Of Life so I think we should not use it....)
The another package manager you mentioned is Anaconda isn't it?
I also use this package manager so I am thinking of it, but I am not familiar with the way of making packaging system for anaconda repository.

According to [PEP621](https://peps.python.org/pep-0621/), the python packaging procedure is been integrated into one file : pyproject.toml, and I think it is required to follow the PyPA rules in order to establish the important position in the at least fusion OSS community.

Anyway, I will try to look for how to resolve the dependencies' issue.

ssilburn, 2023-08-02:
I'm now finally going to close this issue because in release v2.11 Calcam now uses `pyproject.toml` and does things in a more pep-compliant way. As on the other issue, please can you open a new issue for anaconda distribution?

closed 2023-08-02 by ssilburn