euratom-software/calcam #90: Black / flickering views with VTK 9 + PyQt6 on Windows

ssilburn, 2022-07-17:
Using VTK 9 & PyQt6 on Windows causes black / flackery graphics. Doesn't seem to happen with either older PyQt or older VTK.

perek42, 2022-07-28:
Same problem on a MacBook Pro (13-inch, M1, 2020) mac running macOS 10.4 and python 3.9.13
pip list 
Package                 Version
----------------------- ---------
absl-py                 1.1.0
aiohttp                 3.8.1
aiosignal               1.2.0
astunparse              1.6.3
async-timeout           4.0.2
attrs                   21.4.0
autobahn                22.6.1
Automat                 20.2.0
brotlipy                0.7.0
cachetools              5.2.0
calcam                  2.9.1
certifi                 2022.6.15
cffi                    1.15.0
charset-normalizer      2.0.12
click                   8.1.3
colorama                0.4.4
conda                   4.13.0
conda-package-handling  1.8.1
constantly              15.1.0
cryptography            37.0.2
cycler                  0.11.0
flatbuffers             1.12
fonttools               4.34.4
frozenlist              1.3.0
gast                    0.4.0
google-auth             2.9.0
google-auth-oauthlib    0.4.6
google-pasta            0.2.0
grpcio                  1.42.0
h5py                    3.6.0
hyperlink               21.0.0
idna                    3.3
importlib-metadata      4.12.0
incremental             21.3.0
joblib                  1.1.0
keras                   2.9.0
Keras-Preprocessing     1.1.2
kiwisolver              1.4.4
libclang                14.0.1
Markdown                3.3.7
matplotlib              3.5.2
multidict               6.0.2
numpy                   1.22.3
oauthlib                3.2.0
opencv-python-headless  4.6.0.66
opt-einsum              3.3.0
packaging               21.3
Pillow                  9.1.1
pip                     22.1.2
protobuf                3.19.4
pyasn1                  0.4.8
pyasn1-modules          0.2.8
pycosat                 0.6.3
pycparser               2.21
pydantic                1.9.1
pyOpenSSL               22.0.0
pyparsing               3.0.9
PyQt6                   6.3.1
PyQt6-Qt6               6.3.1
PyQt6-sip               13.4.0
PySocks                 1.7.1
python-dateutil         2.8.2
PyYAML                  6.0
requests                2.27.1
requests-oauthlib       1.3.1
rsa                     4.8
ruamel-yaml-conda       0.15.80
scipy                   1.8.1
setuptools              62.3.2
six                     1.15.0
tensorboard             2.9.1
tensorboard-data-server 0.6.1
tensorboard-plugin-wit  1.8.1
tensorflow-addons       0.17.1
tensorflow-estimator    2.9.0
tensorflow-macos        2.9.2
tensorflow-metal        0.5.0
termcolor               1.1.0
tqdm                    4.64.0
Twisted                 22.4.0
txaio                   22.2.1
typeguard               2.13.3
typer                   0.4.1
typing_extensions       4.2.0
urllib3                 1.26.9
vtk                     9.0.3
Werkzeug                2.1.2
wheel                   0.37.1
wrapt                   1.14.1
wslink                  1.6.6
yarl                    1.7.2
zipp                    3.8.0
zope.interface          5.4.0

ssilburn, 2022-11-28:
Due to this and some other issues with PyQt6 + VTK9 (causes complete failure of the GUI to open on my Ubuntu machine!), plus because PyQt5 is still widely easily available - for now I've just removed support for PyQt6 from Calcam. In the current state of compatibility with VTK it's causing many more problems than it's worth to be compatible with the latest version.

ssilburn, 2025-03-17:
Testing this again now, with Calcam 2.15.dev, PyQt6 and VTKs 9.2.6 and 9.41 - it looks like these libraries have learned to play nicely with each other and the problem is gone, at least on the Intel integrated graphics machine I had it on before. So unless there are more problems found I propose to re-enable PyQt6 support and make PyQt6 the default requirement for calcam 2.15.

closed 2025-05-08 by ssilburn