euratom-software/calcam #100: Image offsets missing

perek42, 2023-03-13:
Hi,

I've seen this problem before and am unsure where it comes from, but here is a hotfix:

change the line in core.py
newim = self.imsource.get_image_function(**imload_options)
to:
newim = self.imsource.get_image_function(**imload_options, offset_x=0, offset_y=0)

Cheers, 
Artur

DIAGNOSTIC INFORMATION
----------------------

Platform:       darwin
Python version: 3.10.10 (main, Feb 16 2023, 02:49:39) [Clang 14.0.0 (clang-1400.0.29.202)]
Calcam version: 2.10.0
VTK version:    9.2.6
OpenCV version: 4.6.0
PyQt version:   5.15.7

Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.10/site-packages/calcam-2.10.0-py3.10.egg/calcam/gui/core.py", line 449, in load_image
    newim = self.imsource.get_image_function(**imload_options)
TypeError: get_image() missing 2 required positional arguments: 'offset_x' and 'offset_y'

martimzurita, 2023-05-25:
Thanks a lot, Artur! I wish a could thank you in person!

ssilburn, 2023-08-02:
I'm going to close this because I'm pretty sure it should be fixed in v2.11 which is now released.

closed 2023-08-02 by ssilburn