euratom-software/calcam #101: GUI not starting properly on Windows 10

iabramovic, 2023-05-04:
After installing calcam I am able to get the GUI showing the windows with the calcam logo and on the right options 
calibration tool point fitting 
calibration tool manual alignment 
etc

clicking on any of the listed options does not work, I am stuck at the initial window 

any tips on how to fix this? thanks!

ssilburn, 2023-07-31:
Hi, really sorry for the lack of response!

If it's still relevant, please could you try the following commands at a Python prompt - this should try to open the GUI window, so should reveal an error message to show what's going wrong:

`import calcam`
 `calcam.gui.open_window(calcam.gui.FittingCalib)`

ssilburn, 2023-11-12:
Hi, is this still relevant / happening? If I don't hear anything I'll close the issue. In the meantime, the error handling early on in the GUI initialisation has been improved to hopefully prevent silent failures like this.

closed 2024-11-04 by ssilburn