euratom-software/calcam #102: How should a point pair csv file look like?

Csega, 2023-06-02:
Dear Developers,

thanks for this spatial calibration software! I find it very useful. However, I could not find in the documentation, how to format the calibration point pairs CSV file? I already figured out from the source code, that the header line should start with "Machine X". I guess, the next parameters are "Machine Y, Machine Z", and probably "Image X, Image Y"? This solution does not work and I could not find further hints on what is the good layout. Could you help me with that, please?

 It would also be great if one could export the point pairs from a working calibration, not just use them from the other calibration file (.ccc).

Cheers,
Csega

ssilburn, 2023-07-31:
Hi Csega,

Sorry for the lack of responsiveness! And for the lack of documentation about the file formats - this needs to be done! If you want direct access to the point pairs from a calibration, you should be able to open .ccc files with an archive manager / zip program, and find inside it the pointpairs.csv file. I've also attached an example of one to show the format. 
[pointpairs_example.csv](https://github.com/euratom-software/calcam/files/12217791/pointpairs_example.csv)
I'll leave this issue open with the action on me to document the file formats in the documentation!

ssilburn, 2023-08-02:
Hi,

There's now a page added to the documentation [here](https://euratom-software.github.io/calcam/html/dev_fileformats.html) which gives at least basic information about the formats, so I'll now close this issue. Please open another one if you have any more problems working with the files or the documentation isn't sufficient.

closed 2023-08-02 by ssilburn