euratom-software/calcam #103: pip installation doesn't work

munechika-koyo, 2023-06-28:
The calcam installation with pip command does not successfully done according to the following logs:
<details>
<summary>
❯ pip install -v calcam
</summary>

```bash
Using pip 23.1.2 from /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages/pip (python 3.9)
Collecting calcam
  Using cached calcam-2.10.0.tar.gz (485 kB)
  Running command python setup.py egg_info

  ***************************************************************
                  Calcam v2.10.0 Setup Script
  ***************************************************************

  running egg_info
  creating /tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info
  writing /tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/entry_points.txt
  writing requirements to /tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/SOURCES.txt'
  adding license file 'LICENCE.txt'
  writing manifest file '/tmp/pip-pip-egg-info-7ic6v1je/calcam.egg-info/SOURCES.txt'

  ***************************************************************

  Preparing metadata (setup.py) ... done
Requirement already satisfied: scipy in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from calcam) (1.11.0)
Requirement already satisfied: matplotlib in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from calcam) (3.5.3)
Requirement already satisfied: cycler>=0.10 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (0.11.0)
Requirement already satisfied: fonttools>=4.22.0 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (4.40.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (1.4.4)
Requirement already satisfied: numpy>=1.17 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (1.25.0)
Requirement already satisfied: packaging>=20.0 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (23.1)
Requirement already satisfied: pillow>=6.2.0 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (9.5.0)
Requirement already satisfied: pyparsing>=2.2.1 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (3.1.0)
Requirement already satisfied: python-dateutil>=2.7 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from matplotlib->calcam) (2.8.2)
Requirement already satisfied: six>=1.5 in /home/koyo/.conda/envs/cherab-phix-dev/lib/python3.9/site-packages (from python-dateutil>=2.7->matplotlib->calcam) (1.16.0)
Building wheels for collected packages: calcam
  Running command python setup.py bdist_wheel
  Traceback (most recent call last):
    File "<string>", line 2, in <module>
    File "<pip-setuptools-caller>", line 34, in <module>
    File "/tmp/pip-install-muvhhwto/calcam_0af0b89610a143aeaeb987bfed463aa7/setup.py", line 42, in <module>
      raise Exception("Calcam currently cannot be distributed as wheels due to having important install-time logic in setup.py.")
  Exception: Calcam currently cannot be distributed as wheels due to having important install-time logic in setup.py.
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /home/koyo/.conda/envs/cherab-phix-dev/bin/python3.9 -u -c '
  exec(compile('"'"''"'"''"'"'
  # This is <pip-setuptools-caller> -- a caller that pip uses to run setup.py
  #
  # - It imports setuptools before invoking setup.py, to enable projects that directly
  #   import from `distutils.core` to work with newer packaging standards.
  # - It provides a clear error message when setuptools is not installed.
  # - It sets `sys.argv[0]` to the underlying `setup.py`, when invoking `setup.py` so
  #   setuptools doesn'"'"'t think the script is `-c`. This avoids the following warning:
  #     manifest_maker: standard file '"'"'-c'"'"' not found".
  # - It generates a shim setup.py, for handling setup.cfg-only projects.
  import os, sys, tokenize
  
  try:
      import setuptools
  except ImportError as error:
      print(
          "ERROR: Can not execute `setup.py` since setuptools is not available in "
          "the build environment.",
          file=sys.stderr,
      )
      sys.exit(1)
  
  __file__ = %r
  sys.argv[0] = __file__
  
  if os.path.exists(__file__):
      filename = __file__
      with tokenize.open(__file__) as f:
          setup_py_code = f.read()
  else:
      filename = "<auto-generated setuptools caller>"
      setup_py_code = "from setuptools import setup; setup()"
  
  exec(compile(setup_py_code, filename, "exec"))
  '"'"''"'"''"'"' % ('"'"'/tmp/pip-install-muvhhwto/calcam_0af0b89610a143aeaeb987bfed463aa7/setup.py'"'"',), "<pip-setuptools-caller>", "exec"))' bdist_wheel -d /tmp/pip-wheel-yc56vnav
  cwd: /tmp/pip-install-muvhhwto/calcam_0af0b89610a143aeaeb987bfed463aa7/
  Building wheel for calcam (setup.py) ... error
  ERROR: Failed building wheel for calcam
  Running setup.py clean for calcam
  Running command python setup.py clean

  ***************************************************************
                  Calcam v2.10.0 Setup Script
  ***************************************************************

  running clean
  'build/lib' does not exist -- can't clean it
  'build/bdist.linux-x86_64' does not exist -- can't clean it
  'build/scripts-3.9' does not exist -- can't clean it

  ***************************************************************

Failed to build calcam
ERROR: Could not build wheels for calcam, which is required to install pyproject.toml-based projects
```
</details>

This seems to be caused by the prohibition of `bdist_wheel`.
I guess the latest `pip` forces to build the wheel dists even though using the legacy-installation without `pyproject.toml`.
Anyway we should care this issue for beginning users.

munechika-koyo, 2023-06-30:
This issue seems to be caused by the deprecation of the fallback source installation by `pip`.
If I changed the `pip` version to `v23.0`, then `calcam` was able to be installed.
Here is the logs:

<details>
<summary>
❯ pip install -v calcam
</summary>

```bash
Using pip 23.0 from /home/koyo/.conda/envs/test/lib/python3.9/site-packages/pip (python 3.9)
Collecting calcam
  Using cached calcam-2.10.0.tar.gz (485 kB)
  Running command python setup.py egg_info

  ***************************************************************
                  Calcam v2.10.0 Setup Script
  ***************************************************************

  running egg_info
  creating /tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info
  writing /tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/entry_points.txt
  writing requirements to /tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/SOURCES.txt'
  adding license file 'LICENCE.txt'
  writing manifest file '/tmp/pip-pip-egg-info-np0ug4q8/calcam.egg-info/SOURCES.txt'

  ***************************************************************

  Preparing metadata (setup.py) ... done
Requirement already satisfied: scipy in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from calcam) (1.10.1)
Requirement already satisfied: matplotlib in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from calcam) (3.7.1)
Requirement already satisfied: kiwisolver>=1.0.1 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (1.4.4)
Requirement already satisfied: python-dateutil>=2.7 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (2.8.2)
Requirement already satisfied: pillow>=6.2.0 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (9.4.0)
Requirement already satisfied: importlib-resources>=3.2.0 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (5.2.0)
Requirement already satisfied: contourpy>=1.0.1 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (1.0.5)
Requirement already satisfied: packaging>=20.0 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (23.0)
Requirement already satisfied: pyparsing>=2.3.1 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (3.0.9)
Requirement already satisfied: cycler>=0.10 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (0.11.0)
Requirement already satisfied: fonttools>=4.22.0 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (4.25.0)
Requirement already satisfied: numpy>=1.20 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from matplotlib->calcam) (1.25.0)
Requirement already satisfied: zipp>=3.1.0 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from importlib-resources>=3.2.0->matplotlib->calcam) (3.11.0)
Requirement already satisfied: six>=1.5 in /home/koyo/.conda/envs/test/lib/python3.9/site-packages (from python-dateutil>=2.7->matplotlib->calcam) (1.16.0)
Building wheels for collected packages: calcam
  Running command python setup.py bdist_wheel
  Traceback (most recent call last):
    File "<string>", line 2, in <module>
    File "<pip-setuptools-caller>", line 34, in <module>
    File "/tmp/pip-install-p67kgg5i/calcam_074e0269596b41719cef593928bce41a/setup.py", line 42, in <module>
      raise Exception("Calcam currently cannot be distributed as wheels due to having important install-time logic in setup.py.")
  Exception: Calcam currently cannot be distributed as wheels due to having important install-time logic in setup.py.
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /home/koyo/.conda/envs/test/bin/python3.9 -u -c '
  exec(compile('"'"''"'"''"'"'
  # This is <pip-setuptools-caller> -- a caller that pip uses to run setup.py
  #
  # - It imports setuptools before invoking setup.py, to enable projects that directly
  #   import from `distutils.core` to work with newer packaging standards.
  # - It provides a clear error message when setuptools is not installed.
[... 245 lines cut ...]
  running install_scripts
  Installing calcam script to /home/koyo/.conda/envs/test/bin
  writing list of installed files to '/tmp/pip-record-9sy2wwif/install-record.txt'

  ***************************************************************

  Calcam installation complete.

  It can be imported as a Python module with "import calcam"

  The GUI can be launched using the executable:
  /home/koyo/.conda/envs/test/bin/calcam
  or just by typing "calcam" at a terminal.

  ***************************************************************

  Running setup.py install for calcam ... done
  DEPRECATION: calcam was installed using the legacy 'setup.py install' method, because a wheel could not be built for it. pip 23.1 will enforce this behaviour change. A possible replacement is to fix the wheel build issue reported above. Discussion can be found at https://github.com/pypa/pip/issues/8368
Successfully installed calcam-2.10.0
```

</details>

According to the log, legacy-install (i.e. without building wheels?) has deprecated at version 23.1.
Therefore, we should check the pip version before an installation.

ssilburn, 2023-07-31:
OK, no more hiding from the install method issue then!

There still is no way to have flexible dependencies in the official way of Python packaging, but I think now PyQt5, OpenCV and VTK are available *enough* on different platforms that it's safe enough to abandon setup.py and use the modern way of doing things.

munechika-koyo, 2023-08-01:
I was looking forward to hearing the good news :+1: 
Regarding the other topics, I would like to recommend that `calcam` should be available for an anaconda user by registering it in a `conda-forge` channel. Unfortunately, I don't really understand how to conduct that, but what do you think about my idea?

ssilburn, 2023-08-02:
Yeah sounds reasonable to investigate how to also package & release it for conda, I'm not familiar with it either because I don't / haven't used conda so will need to find out how. Please could you open a new separate issue for that? I'll now close this one because the pip compatibility is fixed in release 2.11.

closed 2023-08-02 by ssilburn