euratom-software/calcam #106: Requesting Support for Conda Packaging

munechika-koyo, 2023-08-02:
Hi,

As discussed at #83 and #103, we should allow users to install `calcam` with `conda`.
Especially, we need to distribute it through `conda-forge` channel, and understand how to conduct  it.

ssilburn, 2023-10-25:
Hi,

Finally had time to have a look at this! I've now added a Conda build recipe on the master branch, which works for build & install for me on the latest Anaconda on Ubuntu.

HOWEVER - when trying to use Calcam it fails / crashes, it looks like because VTK is broken on anaconda, at least on my platform. Error message is:

"libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)"

I don't know how platform-specific this is, but if VTK is not working on Anaconda generally, it seems to not make much sense to distribute it that way.

If you can, please could you try conda-build and conda-install using the recipe now in the conda/ directory on the master branch, and see if it works?

If the problem I have is specific and on most platforms it works, I will add a GitHub action to build the conda package and upload it to Anaconda.org automatically on release (like it does now for PyPi).

ssilburn, 2023-10-26:
(previous message deleted due to new developments)

I think I've now got a working conda build recipe included. I'll add a GitHub action for building & publishing the anaconda package at the next release, not through `conda-forge` at first but at least through a public channel.

munechika-koyo, 2023-10-27:
Thank you for your updates!
I will still wait for `conda-forge` support!

ssilburn, 2023-11-12:
OK, 2.12 released and is available on Anaconda.org; install instructions added to the Calcam documentation. Appears to be working correctly; I'll close this issue now.

closed 2023-11-12 by ssilburn

munechika-koyo, 2023-11-13:
Thank you for making `calcam` available on Anaconda.org, which may be the `default` channel on conda.
However, is `conda-forge` compatibility out of scoop?
This is because it cannot be found when one uses [Miniforge](https://github.com/conda-forge/miniforge) which limits only `conda-forge` channel.
So, I would appreciate it if you would handle this issue in the future.