euratom-software/calcam #112: Camera Calibration Fitting is successfull but the CAD view is flipped

GabrielePartesotti, 2023-09-14:
Hi, 

I have a pinhole camera that is looking inside a machine. I already know the pinhole coordinates, the pixel coordinates, and the intersections of each pixel line of sight with the CAD model of the machine. I just want to generate a picture of the CAD model projected on my camera view.  
I load the CAD model and the point pairs. I perform the fit and it seems like all points are matching nicely. 
[image]
However, when I set the CAD view to match the fit the image is totally wrong and it's facing away from the machine. I think it is actually rotated by roughly -180° since if I drag it manually I can place it more or less where it should be.
[image]
What I would like to get is something similar to the figure below, in which I manually adjusted the view:
[image]

I can also provide the CAD models and the points if that helps. 
My calcam version is 2.11.0. on  a Ubuntu 20.04 VM.

ssilburn, 2023-10-18:
Hi, really sorry for the slow reply! 

From the images you posted, it looks like the image you're fitting is flipped vertically? (by looking at the positions of the unobstructed port and the one partly hidden by the divertor plate). My guess is that's why Calcam has had to flip the camera 180 degrees to get the model to fit - it only works properly if the image is not horizontally or vertically flipped compared to reality (If it's convenient, the flip can be added inside Calcam with its geometrical transofmrations, then you can choose when generating subsequent images whether to generate them with or without the flip).

Another option, since you already know the pinhole position and field of view, would be to set it up using the "Virtual Calibration" editor, where you can directly enter the pinhole position, pixel dimensions and field of view.

Again sorry for being so slow...maybe you already solved it or gave up by now! Let me know if this helps or you have more questions!

GabrielePartesotti, 2023-10-23:
Hi,

Yes I managed to solve it some time ago. As far as I remember I had to flip the order in which I was inputting the calibration points. After fixing this, for some reason the option to match the CAD view to the calibrated one was still not exact, but the rendered image was.

Best,

Gabriele Partesotti

ssilburn, 2023-10-23:
Ah OK, great! Yes the button to set the view to match the calibration makes some compromises because it can't include lens distortion and also the window aspect ratio is usually different from the camera, so it's not exact - but the image rendering is.

Since this doesn't appear to be a problem with the code I'll close this issue.

closed 2023-10-23 by ssilburn