euratom-software/calcam #116: Uploading `conda-forge` channel

munechika-koyo, 2023-12-07:
Hi!

I am trying to add `calcam` to `conda-forge` channel following [this](https://conda-forge.org/docs/maintainer/adding_pkgs.html) instruction and posted the PR to `conda-forge/staged-recipes/` repository: https://github.com/conda-forge/staged-recipes/pull/24702

If you are welcome this addressing, could I have it reviewed by `conda-forge` maintainer or accomplish it?
I think there is room for who is included as a recipe's maintainer.

Thank you in advance.

munechika-koyo, 2023-12-25:
I noticed that `calcam` with only python 3.11 are uploaded in the Anaconda calcam channel probably because of the timeout from #114?
`Conda-forge` registering should resolve this issue as well.
Additionally, the reviewer asked me why `libstdcxx-ng` is required [here](https://github.com/conda-forge/staged-recipes/pull/24702#discussion_r1421028655). Could someone please answer this?

ssilburn, 2024-02-13:
Hi!

Really sorry I didn't reply to this for so long!

The reason for the `libstdcxx-ng` `run_constrained` requirement is because when I was testing the recipe (on Ubuntu), I found that with the version of `libstdcxx-ng` from the main channel, which is v11.2, VTK is broken! When I try to start any calcam GUI which uses VTK, I get a crash with error:

```
The read operation timed out
libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

Updating `libstdcxx-ng` to the version from `conda-forge` which is v13.2 fixes the problem. So the requirement on `libstdcxx-ng` is to force the update of this library at install time.

I just did a new release of Calcam to v2.13 - it also introduces a couple of new dependencies. I'm happy for you to update your recipe and reply to the reviewer with this info if you want, or I also prepared one at https://github.com/ssilburn/conda-forge-staged-recipes which I could do a pull request from - whichever you prefer!

munechika-koyo, 2024-02-14:
Thank you for responding to my comment, and congratulations on the new release!
I have also updated my recipe and submitted a [PR](https://github.com/conda-forge/staged-recipes/pull/24702).

I kindly request that you take a look at the PR and leave a comment as it requires the maintainer's input. 
Thank you in advance for your time and assistance

munechika-koyo, 2024-08-12:
I noticed that in `conda-forge` channel latest `triangle` package is not distributed (only old versions exist):
https://prefix.dev/channels/conda-forge/packages/triangle.
It might be required to ask triangle developers to contribute to it.

ssilburn, 2024-11-05:
Looks like triangle is now being updated on conda forge again!

Meanwhile, Calcam is now successfully being auto-updated on the "calcam" Anaconda channel at least

munechika-koyo, 2024-11-06:
That's good news!
Congrats on a new release, and I would like to address conda-forge work again.

munechika-koyo, 2025-01-20:
Done at https://github.com/conda-forge/staged-recipes/pull/24702

closed 2025-01-20 by munechika-koyo