euratom-software/calcam #122: ERROR: Could not build wheels for triangle

EJCT22, 2024-04-09:
I have this error when trying to install calcam using pip:

triangle/core.c:196:12: fatal error: 'longintrepr.h' file not found
        #include "longintrepr.h"
                 ^~~~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for triangle
Successfully built calcam
Failed to build triangle
ERROR: Could not build wheels for triangle, which is required to install pyproject.toml-based projects

I tried to create pyenv to solve this but it still did not work.

ssilburn, 2024-04-10:
Hi, please can you let me know what OS & Python version this is in?

ssilburn, 2024-04-10:
Great, glad you got it working! I'll leave this issue open and put some more explanation here because it might be useful for other users on similar platforms. Are you on an Apple silicon machine by any chance?

Thanks a lot to @pixelifytica for helping me investigate this. As far as we can tell it's probably is specific to Apple silicon machines  with Python 3.9+ ,  so easiest solutions are either to do as @EJCT22 did and install Calcam with `--no-deps` then install the other missing modules "by hand", or to use Python 3.9 or older on Apple silicon machines.

It looks like this happens because the OSX wheels for triangle on PyPi are only for x86-64, so on Apple silicon pip will try to build triangle from source. However, changes to CPython after v3.9 have broken triangle's build against them, and with Python 3.10 or newer this results in exactly the error above.

EJCT22, 2024-04-11:
Thanks for the explanation. I think that’s why it didn’t work when I used venv with python11. Yes, I am using Apple silicon.

ssilburn, 2026-09-15:
Noting this as an update here: according to this page - [pip install git+https://github.com/drufat/triangle.git](https://blog.rtwilson.com/how-to-install-the-python-triangle-package-on-an-apple-silicon-mac/)

The version of Triangle on GitHub should install OK: `pip install git+https://github.com/drufat/triangle.git`

However I won't update the Calcam dependency to this because it then breaks on other platforms (e.g. Windows if MS VC++14 is not installed) so it's no better overall.

Worryingly, it looks like triangle might not be actively maintained on PyPi now, so finding a way to drop this dependency is probably the best thing to do.

ssilburn, 2026-10-11:
The situation with triangle actually seems to be getting worse: the PyPi package appears to be stale and there is a proliferation of others named "triangle2" and "triangle3", and the GitHub repo has not been updated in some time. So for long term reliability, Calcam will change to using the meshpy package to access triangle, which appears to be actively maintained and produces very similar results.

This is now changed on the master branch and will come in to effect in the next release.

ssilburn, 2026-10-15:
Closed at v2.19 which has dropped the dependency on `triangle` in favour using `meshpy`

closed 2026-10-15 by ssilburn