euratom-software/calcam #127: Saving RayData with non-zero binning

rdoyle45, 2024-10-01:
The below issue arises when trying to save raydata when the calcam.raycast_sightlines() was run with binning != 0. I have looked on SciPy docs and it appears use of the assignValue() method is appropriate but I am not sure why it does not work.

`In [1]: import calcam

In [2]: cam_calib = calcam.Calibration('../CIS_Calibrations/MU03_calcam_calib.ccc')

In [3]: mastu_machine = calcam.CADModel('MAST Upgrade')
Extracting CAD model...

In [4]: raydata = calcam.raycast_sightlines(cam_calib,mastu_machine, binning=10)

Casting 49.8k rays...
Completed in:       2 sec. 

In [5]: raydata.save('raydata_cropped_sensor_ds10.nc')
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Input In [5], in <cell line: 1>()
----> 1 raydata.save('raydata_cropped_sensor_ds10.nc')

File ~/CIS/calcam/calcam/raycast.py:343, in RayData.save(self, filename)
    340 binning = f.createVariable('Binning','f4',())
    342 if self.binning is not None:
--> 343     binning.assignValue(self.binning)
    344 else:
    345     binning.assignValue(0)

File ~/venv/lib64/python3.9/site-packages/scipy/io/_netcdf.py:943, in netcdf_variable.assignValue(self, value)
    935 if not self.data.flags.writeable:
    936     # Work-around for a bug in NumPy.  Calling itemset() on a read-only
    937     # memory-mapped array causes a seg. fault.
    938     # See NumPy ticket #1622, and SciPy ticket #1202.
    939     # This check for `writeable` can be removed when the oldest version
    940     # of NumPy still supported by scipy contains the fix for #1622.
    941     raise RuntimeError("variable is not writeable")
--> 943 self.data[:] = value

IndexError: too many indices for array: array is 0-dimensional, but 1 were indexed
`

ssilburn, 2024-11-04:
Hi Rhys,

Sorry it took me so long to look at this but glad you found a workaround! 

It looks like this is an issue that has appeared with SciPy 1.12 or newer. I can't understand either from their documentation why this would have happened, but I can't find a better solution than yours, and your solution seems to still work fine with older SciPy versions, so I'll adopt what you posted here on to master now and it'll be in the next release.

ssilburn, 2024-11-05:
Now included in release v2.14

closed 2024-11-05 by ssilburn